earlephilhower/BackgroundAudio #1: AudioOutputBase.h is Missing

MasonStooksbury, 2024-12-19:
Not sure if this is a me issue but it seems that the AudioOutputBase.h header file is missing?

I'm using a Pi Pico W and have been using the arduino-pico library. Haven't had any issues with anything else but this being really knew, just wanted to make sure I'm not missing something obvious. I installed this manually by downloading the .zip of the latest release and using the "Sketch -> Include Library" method in the Arduino IDE. 

This is the error I get when doing your example and attempting to compile the project

[image]

earlephilhower, 2024-12-19:
You need the latest 4.4.0 or `git master` release of arduino-pico.  I just recently refactored the audio output devices to a common interface class and it was part of [that release](https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/AudioOutputBase.h) yesterday.

MasonStooksbury, 2024-12-19:
Ah, yep, that's what I was missing. Removing the old installation entirely, adding the new board manager URL for the 4.4.0 release and installing that solved almost all the issues.

Last thing it's complaining about now is that `filebuff` isn't a thing. Looks like in your example we don't instantiate or assign anything for `filebuff`. Is this just an array that I need to direct the file bytes to?

earlephilhower, 2024-12-19:
The example reads SD into filebytes, then sends it to the decoder. as an opaque set of bytes  Just a temp buffer that's 1 sector in size so hopefully SDFat reads it efficiently.

MasonStooksbury, 2024-12-19:
Makes sense. Adding a `uint8_t filebuff [512];` before setup() seems to have compiled. I should be able to figure out anything else from here. Thanks so much for your help!

deladriere, 2025-01-26:
I have also the error " AudioOutputBase.h is missing " using Plattformio
Here is my platformio.ini
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower

lib_deps =
 https://github.com/earlephilhower/BackgroundAudio.git
```
is this the correct setup ?

deladriere, 2025-01-27:
Thanks! But it still gives the error message AudioOutputBase.h: No such file or directory
the PIO version is the latest
```
You're up-to-date!
PlatformIO 6.1.16 is currently the newest version available.
````
and I updated the packages too

````
pio pkg list
Resolving pico dependencies...
Platform raspberrypi @ 1.15.0+sha.39b9039 (required: git+https://github.com/maxgerhardt/platform-raspberrypi.git)
├── framework-arduinopico @ 1.40300.0+sha.e16c459 (required: git+https://github.com/earlephilhower/arduino-pico.git#e16c459598b86ce67d8b83ac5b1384dcb8c982ae)
├── tool-picotool-rp2040-earlephilhower @ 5.140200.240929 (required: https://github.com/earlephilhower/pico-quick-toolchain/releases/download/4.0.1/aarch64-apple-darwin20.4.picotool-8a9af99.240929.tar.gz)
└── toolchain-gccarmnoneeabi @ 1.90301.200702 (required: platformio/toolchain-gccarmnoneeabi @ >=1.90201.0,<1.90302.0)

Libraries
└── BackgroundAudio @ 1.3.0+sha.2b6d490 (required: git+https://github.com/earlephilhower/BackgroundAudio.git)
```

earlephilhower, 2025-01-27:
You've reinstalled an old version of the core, 4.0.3.  You need to tell PIO to update that to the latest one.  You can use Git version instead, check the docs.