earlephilhower/BackgroundAudio #3: Burst of white noise when BeepWAV example begins

DanEfran, 2024-12-22:
On a KB2040 board, I'm hearing a short (about 1 second) burst of "static" when the BeepWAV example calls BMP.begin(). It sounds as though maybe it's playing uninitialized buffers? 

Testing with other wav data (and messing about with flush(), delay(), etc.) I was able to get a case where the noise appeared at the begin() call and also at the start of the first write(wav...) call -- but only the first one -- again suggesting junk in the buffers before the real data shows up.

The BeepWAV example otherwise works properly; other forms of PWM sound playback (e.g. ToneHappyBirthday example, CircuitPython mp3 playback) also seem to work properly.

I briefly glanced over the library code, but I didn't see anything obviously amiss. However, BMP.begin() works its way down through several classes, and there seem to be linked lists involved...I didn't trace through every last detail. Since the problem appears right at the outset in the example sketch, I have to suspect the library is somehow not zeroing everything it should be.

By the way, it's great to see this library show up exactly the week I was wishing for it! Thanks!!

earlephilhower, 2024-12-22:
Glad it's useful!  I just got sick of looking at how wasteful the ESP8266Audio library turned out.  It can do lots of stuff, but half he time it's just jumping around vtables and thrashing instruction caches and not actually DSPing.

What output were you running, PWM or I2S?  And after it starts up things are normal, right?  I normally use I2S and can't say I've heard that but on some DAC chips it takes a few frames before it syncs up I think.

Anyway, the Happy B'day example uses PIO tone generators, not PWM or I2S, so that should be completely unrelated to any code this hits.

Looking through the WAV code I can't see any buffers that could be sent out with random data.  At startup there could be underflow, but the `goto` branch (hey, they're legal in C++ and avoid cut-n-paste while being fast!) will write 0s to out until it's full.

That would indicate it's not the library here that's got a startup problem, but the Arduino PWM/I2S or AudioBufferDMA one.  I'll check in the core for anything obvious...

DanEfran, 2024-12-22:
After more testing, I think there may be something more subtle going on. I'm using an audio amplifier, which I thought I had ruled out as part of the issue (since it plays other sounds successfully through the same pin). But there's a 'shutdown' pin on the amplifier, and if I use that to mute the output I can work around the problem:

  digitalWrite(PIN_MUTE, LOW);
  BMP.begin();
  delay(29);
  digitalWrite(PIN_MUTE, HIGH);

I figured I could hide the noise by muting the amp for a moment, and I could tune the delay to just long enough to cover it. But what I found did not fit my expectations: if I change the delay(29) to delay(28), the noise comes back -- all of it! Not just a click, but the entire approximately one second of noise.

That suggests to me that the noise is maybe actually generated by the amplifier board freaking out. (It's adafruit's PAM8302 board, http://adafru.it/2130, if that helps.) It's receiving something from the begin() call that it can't handle -- possibly not a full second of junk data after all, maybe just some transient signal that makes it send its own second of junk in response.

That suggests a somewhat different line of inquiry into this issue; I hope it helps. I'm not sure what would provoke such a response from the amp; I'm still not sure what's really going on. It does still seem like something specific to this library's startup process: my understanding is that the CircuitPython mp3 PWM playback example I tested used essentially the same PWM technique (I think?) and that seemed to work. As you said, the ToneHappyBirthday example uses a different technique, but sound through the same pin and amp combination work fine. So it's not a general amp startup problem, and it's not a sound playback problem because this library goes on to work fine. It's some issue with the sequence of events in begin(). Right around 28.2 milliseconds into the process you're doing something the amp hates. That's my current guess.

(Meanwhile I have a workaround, at least, and it even uses less startup time than I thought it would. I figured I'd have to wait out the whole second.)

earlephilhower, 2024-12-22:
You can't really run PWM through an amplifier without low pass filtering it or you're going to have some very ugly effects.  It's not analog, it's just full on/full off repeatedly with a specific duty cycle at a very high frequency.  I'm kind of surprised you can get anything intelligible out with it.  Try the usual RC network and see if that helps.  You should also check the Vin for the amp.  This is sending 0-3.3V, many amps want a 0-1V swing.

Tone runs a square wave at 1-4khz frequency, not a variable width one at 10MHz, so Tone might work since there really aren't those high order harmonics being used.

I can't reproduce a second of noise at all in any of my tests here directly driving a small earbud.  There's some low-level noise, but that's a function of PWM not having all that many bits and not 1s of screeching or anything.

earlephilhower, 2024-12-22:
Some more checking confirms good data is going out to PWM (using GDB).  It's 0s on the beginning until the WAV data gets in and then it's the beep.  There is a glitch for ~1/10th of a second on first startup because finding the clock ratio is very slow and disables IRQs for a search of 65535 possible clocking options using floating point math (!!!).  After the initial clock setting this call is instantaneous and no static comes out.  Again, only the initial clock setting gives this and afterwards we're all clean (and in fact no bad data ever gets written to PWM DMA...it's just the DMA engine can't be services in enough time while doing that initial calc).

earlephilhower, 2024-12-22:
With the fix in arduino-pico, I'm not getting any static at all on a directly coupled speaker.  So this is related to driving an analog amp off of a pwm signal.  You might consider just driving a jFET or something off the PWM signal to actually amplify things, or an RC network between PWM output and the amp.

DanEfran, 2024-12-23:
I do have a bit of a low-pass filter before the amplifier -- a poor-man's DAC, just a few capacitors and resistors, but I'm using what appears to be the RPi-recommended circuit for such applications. 

As it happens, the sound quality is still ok even without that, just direct pin to amp, which surprised me. As you say, it shouldn't really work that way.

In any case, I am indeed using a theoretically suitable smoothing circuit for these tests. So I don't think that's much of a factor here -- after all, once it gets going, the sound from the library is fine.

It sounds like the culprit has to be that clock-ratio search on startup. It's exactly the kind of low-level one-time code I was hoping to find at the scene of the crime. I don't know what that process does to the pin's output during that 1/10 sec, but apparently the amp reacts by making noise. (Either I'm hearing the scan itself happening, or the pin is floating and I'm hearing amplified floating pin noise, or the scan puts out a signal that, even through a capacitor, puts the amp into some kind of feedback frenzy unrelated to the input as audio...? I don't know.)

Sounds like you've been able to skip the scan in most cases, which I imagine fixes the problem. Probably. I suppose the remaining handful of clock tests could still be enough weirdness to freak out the amp.

I'll test the fix when I get a chance, but it looks like that means installing arduino-pico the git way, so I probably won't get to it today. Meanwhile, though, thanks for the instantaneous response!

DanEfran, 2024-12-23:
It’s going to be a few days before I can get back to these tests. I’d like to leave this issue open for now, until I can confirm that the fix resolves the issue fully or not. 

I ran another round of tests (on the released version of the library, without the fix) with my alternate WAV data and again was able to get multiple bursts of noise, one on calling begin() and one at the start of the first playback.  It’s kind of weird. This symptom fits better with the junk buffers theory and is hard to reconcile with the clock scan theory. Or maybe it’s something about the clock ratio, or memory access…I’m really puzzled. I don’t think my WAV data is bad, since it plays fine subsequently. 

So I wonder if there’s more to this, or several different things going on. At this point I don’t want to close this issue until I see the fix in action. Sorry it won’t be immediate.

(By the way, I also think I’m seeing the library ignore repeated write() calls (after the first) unless separated by a delay(). Is that expected behavior? I want to pin this down with more tests before I open another issue (if it rises to that level of importance). I won’t get to those tests immediately either, but I figured I’d mention it  now in case you want to take a look.

earlephilhower, 2024-12-23:
First, please make sure you're running the `git master` Arduino-Picountil I get around to releasing another bump.  I try to leave a week or two between releases (unless things are really hosed)and it's only been 5 days or so at this point.

I think it's more likely the amp is losing its mind when PWM begins outputting.  Are you calling `stop` ever?  No need to, PWM will be driven with silence when no WAV data is left.  I'd also recommend bypassing the amp and just connecting the GPIO outs to a high impedance, mini earbud and listening there.  That's my setup (3 dupont connectors soldered onto a phono jack works a charm) and while I was able to repro the screech on initial startip, once the calculation was optimized it's all good.  It never screeched between Beeps even before the change since there was no sampling freq change.

If you're not running the BEep sample here, please zip up and post what you've got so I can run it locally and check.  If you're running beep, then I can verify locally that it does work straight-thru (even interrupting beeps with other beeps).

> ... I’m seeing the library ignore repeated write() calls

Check the return value (or `availableForWrite()`.  When it returns 0 then there's no buffer space right now.  It's like writing to a disk, you need to check the amount of data actually written because maybe the drive's full. :)  The whole idea of the library is that the main app doesn't need to block while the audio is being decoded, so sometimes you try and write 8kb and there's only 1kb of buffer space so it'll write the 1k and tell you so.  Your app then does its UI, web page handing, etc. and on the next loop() tries to write the remaining 7K.  See the WebRadio example, it does Serial and Web UIs between writes of streamed data.

DanEfran, 2024-12-24:
Sounds reasonable. It’ll be a week or so before I can get back to this project, but at that point I’ll try your fix (via git if you haven’t had another release by then), experiment with calling stop(), and send you a copy of my test data that produced the second noise if your fix turns out not to handle that case. I’ll also try it without the amp. I might poke around a little with an oscilloscope too, but I don’t have a very good one so I might not be able to learn much that way.

The availableForWrite stuff makes sense ; I’ll take a closer look at the examples and I’m sure I can work up a data spooling scheme suitable for my project.

DanEfran, 2024-12-29:
Ok, I ran a few more tests.

- Still on the release version, I tried the TRS jack & dupont connectors output method, and everything sounds about the same as with the amp. So I'm pretty sure the amp is not relevant here -- it seems to be operating as it should.
- Moving to the latest git master version, the startup noise is indeed reduced to a much shorter, more tolerable sound. It sounds like a sort of "double-click" or perhaps a handful of very fast clicks -- presumably this is the sound of the (now shorter) clock rate scan.
- I tried adding a pair of end() / begin() calls to each beep in the example, with delays between. The begin() makes a similar "double-click" each time, and the end() call makes a shorter click, similar to the click I get when uploading a sketch. One certainly wouldn't want to call these more often than necessary -- as I think you were saying, I think one begin() call and no end() calls is best for the typical project.
- All these sounds in the fixed version are much more like the little clicks you'd expect from starting or stopping any audio device ever. So I think I'm ready to call this fixed. (Though, I can't help but wonder if there's a way to keep even these remaining begin() clicks from reaching the pin...is it possible to do the scan with the output decoupled from it somehow? I don't know much about the hardware involved.)

Meanwhile, I think I'm seeing another unexpected result, regarding the data spooling side of things. So I might open another issue here soon, if my next tests seem to confirm a problem that isn't my own fault.

closed 2024-12-29 by DanEfran

earlephilhower, 2024-12-29:
Thanks for the update.  Don't call `end` ever and you won't have pops except at startup which is just the PWM pin going from DC 0V to 1.6V average in 10ns.  :)  The library will pump out silence if there's nothing to decode, automatically.