earlephilhower/BackgroundAudio #9: availableForWrite() doesn't work with ROMBackgroundAudioWAV (perhaps by design, but undocumented?)

DanEfran, 2024-12-30:
I'm not sure if this is a bug report, a feature request, or just a request for a few more sentences of documentation to clarify a subtle point of usage.

The BeepWAV example demonstrates the auto-spooling mode alluded to in the README:
"If all data can live in flash, there is a ROM-based playback object which can feed itself, but only from arrays in flash and not files in LittleFS or `SD. See the examples for more details on this mode." So far so good.

I'm interested in repeated or continuous sounds, so I tried modifying the example to play the sound three times in a row, instead of just once, when the button is pressed: three consecutive identical 'write(wav, sizeof(wav))' calls, with no 'flush()' calls between them.

(It is necessary to use a short, low-quality WAV sample; the data array the example sketch ships with is probably too big for this test. I made a little blip that fits comfortably in the buffer multiple times.)

This test did not produce the result I expected: the sound only plays once, and availableForWrite() always returns 0. Always.

Now if I replace ROMBackgroundAudioWAV with BackgroundAudioWAV, then the test succeeds: the sound plays multiple times as I expected.

```
BackgroundAudioWAV: Sound plays three times
wrote  1644d   avail 1644d   afw 6548d
wrote  1644d   avail 3288d   afw 4904d
wrote  1644d   avail 4932d   afw 3260d

ROMBackgroundAudioWAV: Sound plays once
wrote  1644d   avail 1644d   afw 0d
wrote  1644d   avail 1644d   afw 0d
wrote  1644d   avail 1644d   afw 0d
```

I can't tell if "availableForWrite() always returns 0" and "only one WAV at a time even if there's room for more" are intended behaviors for the "ROM-based playback object which can feed itself" mode, but it was surprising that just swapping out the BackgroundAudioWAV type makes such a difference in the library's behavior, and I didn't see any documentation on this subject. 

Since this mode (if I understand correctly) can accept a WAV larger than its buffer, and will consume it automatically, I can see how that might make availableForWrite() less meaningful, or harder to calculate. Perhaps the number would effectively be negative at times. And perhaps the auto-feeding mode complicates things enough to interfere with the normal mode's capacity for user-driven feeding -- or is meant to replace that completely to avoid complications. (If you can add data beyond the buffer capacity, and add more before the current batch is done, where will it all end? (But I still think maybe it ought to work if there is room.)) 

So I can imagine that these differences in the ROM... class may be deliberate. It's understandable that it might have to be that way, or at least that it's much easier that way -- a trade-off for the convenience of the mode. But none of this is particularly obvious at first; it took me a while to figure out exactly what was going on and (maybe) why.

So again, I'm not sure if this is a bug report, a feature request, or just a request for a few more sentences of documentation to clarify a subtle point of usage.

Meanwhile, it seems that the BackgroundAudioWAV class will do what I wanted, using the same ROM-based data array in non-auto-feeding mode. Is there any reason NOT to use BackgroundAudioWAV in place of ROMBackgroundAudioWAV? Are there other differences I should watch out for?

earlephilhower, 2024-12-30:
I can see you've put a lot of work into this, but it's actually behaving as intended.  AFW on the ROM is always 0 because the act of writing it defines the whole buffer.  You'll want to call `done()` to check if it's completed before writing a new buffer.  It would be possible to add in intelligence to support multiple buffers/loops, but that's not there now.  Probably only 10-20 lines for basic functionality.  Keep a list of work (not `std::list<>` though because that does memory allocates/deallocates which you don't want to do at IRQ time) and when the current buffer empties pop an element off the list.  You could get all fancy and support loops, too, if you're into EDM. :laughing: 

The raison d'etre for the ROMxxx methods is that the buffer is never memcpy'd around.  It's processed in-situ and there is no temporary RAM buffer allocated.  So it's lower CPU/memory pressure and simpler code for one-off sounds.  But otherwise it's the same backend as the normal one, so whichever one works for you.

Maybe AFW should report MAXINT when `done()` and 0 OTW.  But in any case the idea is to send the whole sound down in one gulp so the pointer magic can do its work.  If you try and use a ROM buffer piecemeal it's not going to end up well.

(Also, ROM is kind of a misnomer.  You can also write from PSRAM, so you can load a 6MB MP3 file from SD and do 1 call to play it in its entirety.)

DanEfran, 2024-12-30:
> I can see you've put a lot of work into this, but it's actually behaving as intended. [...] It would be possible to add in intelligence to support multiple buffers/loops, but that's not there now.

That's what I figured. As for the work, I don't like to ask about something like this until I've moved past my own mistakes and misunderstandings and developed a clear picture of the issue and whether something is really wrong.

> You could get all fancy and support loops, too, if you're into EDM. 😆

I am, actually. A sample loop box of some kind is a pretty plausible future project for me. Or a wavetable synth even? On that subject, though...my preliminary tests suggest that I might struggle to achieve seamless, timing-precise loops with these techniques. I'll be looking into this more later, but your advice is welcome on getting samples to play exactly one after another with no gaps. (That's why I'm so interested in writing the next sound before the current one is finished, rather than using 'done()': to avoid any chance of underflow. But it sounds like there are still gaps when I do that?)

> The raison d'etre for the ROMxxx methods is that the buffer is never memcpy'd around. It's processed in-situ and there is no temporary RAM buffer allocated.

Nice! I didn't read the code, so I wasn't sure exactly what was going on under the hood. But that makes perfect sense.

> Maybe AFW should report MAXINT when `done()` and 0 OTW.

That's what I'm asking for, I suppose. It sounds a bit silly, but it would allow code using that function to work with that class like it does with the others. And it self-documents the one-WAV limit, doesn't it? Yeah, I guess that's the fix I'm requesting here.

The other thing that would have helped me most would be just another few sentences in the README or in that example's comments (or both) clarifying that the ROM... class only accepts one WAV at a time. The reason is clear in hindsight, but the behavior was surprising compared to how the other classes work and how the documentation describes them. There just isn't much said about the self-feeding mode, so I didn't know to expect these differences.

earlephilhower, 2024-12-30:
For seamless playback, you'll need to either do the normal `BackgroundAudio` (no ROM) or add in that list of pointers capability.  With normal buffering there's always work to do.  With ROM there'll be a time between when the IRQ sends out the last samples and your app realizes its time for another note.

The WAV state machine should accept a WAV right after the end of another and decoded stream should be without a break or silence between the two.  Of course, switching playback speed is going to be slow so make sure everyone is sampled at the same rate.

DanEfran, 2024-12-30:
Sounds promising. I'll do more careful tests of seamless playback when I get a chance.

The list-of-pointers feature would be nice, to bring that class more in line with how the others behave, but it's another layer of complexity that may not be worth the effort. For now, I'd rather just have its absence documented. (Looks like my projects are moving away from the ROM mode anyway.) But I might revisit the idea later, if I continue to miss it. Like you said, it wouldn't be a lot of code.

earlephilhower, 2025-01-23:
Doc issue fixed by #32

closed 2025-01-23 by earlephilhower