earlephilhower/BackgroundAudio #18: KB2040 needs a slightly wider BackgroundAudioMixer

DanEfran, 2025-01-01:
Yes, I'm back with another audio glitch, but this one is very simple.

On my KB2040 test rig, the new Mixer example crackles a fair amount, like it's not quite decompressing fast enough.

I found that it just needs a tiny bit more slack. Changing the example code from
`BackgroundAudioMixer<512> mixer(pwm, 44100);`
to
`BackgroundAudioMixer<576> mixer(pwm, 44100);`
fixes the problem completely. 

Of course, the necessary buffer might vary by hardware or by audio data; but on my test rig, your example code (and data) needed this adjustment.

Thanks for the mixer class!!  It's exactly what I wanted for Christmas.

earlephilhower, 2025-01-01:
I think it's just an optimizer setting.  Make sure you run at `Tools->Optimize->-O2` or `-O3`.  AAC is 50% more work for the RP2040 vs. MP3, and the piano sample I got is at 48khz, too, which is even more compute intensive per second.  `-Os` is something like 10-15% slower, which jives with your increased buffer size.

DanEfran, 2025-01-01:
Fair enough, I can check on that; I'm probably just using default settings. But it looks like if you bump up the number in the example by just a few dozen bytes, it'll work right out of the box without even having to mess with the optimizer.

closed 2025-01-01 by earlephilhower

DanEfran, 2025-01-01:
Just ran a few more tests. For me, -O3 (with <512>) gives about a 95% improvement, but I still consistently hear about three little pops toward the end of the piano sample. I would still have to set it above 512 for perfect results, it seems.

Whereas even at -Os, changing it to <576> eliminates 100% of the pops. 

I recommend raising the number so the example makes the best impression, regardless of the user's optimization settings -- which alone don't solve the issue for me anyway.

DanEfran, 2025-01-01:
Oh, you got it already. Thanks. Hilarious comment.

earlephilhower, 2025-01-01:
What's interesting on this chip is because of the XIP cache and flash reads being so much slower than on a desktop system, `-O3` can actually be worse than `-O2` or even `-Os` in some cases.  It's faster not to unroll many loops, for example, because reading in the extra code for those unrolled instructions takes longer than the extra 1-cycle loop compares.  In any case, the extra 128 samples @ 44khz is only ~3ms extra audio delay if my math is right so good idea to bump it up!

DanEfran, 2025-01-01:
That's interesting. I never really trust one optimizer setting to be better than another just because it has a higher number -- all kinds of weird cases can crop up, especially when it comes to caches; this chip sounds like a perfect example. 

In these tests, I did get pretty similar sounding results between `-O2` and `-O3`: both better than -Os, but not perfect when using the `<512>` buffer.

commit 245bb4a9 (2025-01-01): Bump up buffer in Mixer example

Fixes #18.  640 should be enough for anyone!