earlephilhower/BackgroundAudio #20: Read WAV files in 16000Hz

pierrotm777, 2025-01-03:
Hello,

I use a Waveshare RP2040 Zero.
I read wav file from a SD card and play it with your BackgroundAudio library with success.
But I have some issue with several wav file.
My goal is to load a wav and play it with loop.
But several don't start, other start and stop and several run very well.

It's my actual code:
```
#include <BackgroundAudio.h>
#include <PWMAudio.h>

#include <SD.h>
#include <vector>
#include <string.h>

// This are GP pins for SPI0 on the Raspberry Pi Pico board, and connect
// to different *board* level pinouts.  Check the PCB while wiring.
// Only certain pins can be used by the SPI hardware, so if you change
// these be sure they are legal or the program will crash.
// See: https://datasheets.raspberrypi.com/picow/PicoW-A4-Pinout.pdf
const int _MISO = 4;  // AKA SPI RX
const int _MOSI = 7;  // AKA SPI TX
const int _CS = 5;
const int _SCK = 6;

PWMAudio audio(0);
// We will make a larger buffer because SD cards can sometime take a long time to read
BackgroundAudioWAV BMP(audio);

// List of all MP3 files in the root directory
std::vector<String> wavList;

// The file we're currently playing
File f;

// Read buffer that's better off not in the stack due to its size
uint8_t filebuff[512];

// Recursively scan the card and make a list of all MP3 files in all dirs
void scanDirectory(const char *dirname) {
  File root = SD.open(dirname);
  while (true) {
    f = root.openNextFile();
    if (!f) {
      break;
    }
    String n = f.name();
    n.toLowerCase();
    String path = dirname;
    path += f.name();
    if (f.isDirectory()) 
    {
      if (f.name()[0] == '.') 
      {
        // Ignore . and ..
        continue;
      }
      String sub = dirname;
      sub += f.name();
      //sub += "/";
      scanDirectory(sub.c_str());
    } 
    else if (strstr(n.c_str(), ".wav")) 
    {
      wavList.push_back(path);
      Serial.printf("ADD: %s\n", path.c_str());
    } else {
      Serial.printf("SKP: %s\n", path.c_str());
    }
    f.close();
  }
  root.close();
}

void setup() 
{
  Serial.begin(115200);
  delay(2000);
  
  // Initialize the SD card
  bool sdInitialized = false;
  if (_MISO == 0 || _MISO == 4 || _MISO == 16) {
    SPI.setRX(_MISO);
    SPI.setTX(_MOSI);
    SPI.setSCK(_SCK);
    sdInitialized = SD.begin(_CS);
    Serial.println("SD OK\r\n");
  }
  else 
  {
    Serial.println(F("ERROR: Unknown SPI Configuration"));
    while (1) {
      delay(1);
    }
  }
  if (!sdInitialized) {
    Serial.println("initialization failed!");
    while (1) 
    {
      delay(1000);
      rp2040.reboot();
    }
  }

  scanDirectory("/");

  // Start the background player
  BMP.begin();
  BMP.setGain(0.2);
}

void loop() {
  // Choose a song from the list if there's no open file
  if (!f) {
    //f = SD.open(wavList[1]);
      f = SD.open("CAT-C32.IDL");
      //f = SD.open("DIESEL7.IDL");
      //f = SD.open("DSL-120.IDL");//start and stop
      //f = SD.open("DSL-180.IDL");//sart and stop
      //f = SD.open("DSL-BIG.IDL");//Ok
      //f = SD.open("DSL-LTL.IDL");
      //f = SD.open("DSL-OLD.IDL");
      //f = SD.open("DSL-TUG.IDL");//
      //f = SD.open("DSL-TURB.IDL");//Ok
      //f = SD.open("DSL-V12.IDL");//Ok
      //f = SD.open("SCAN-250.IDL");
      //f = SD.open("SCAN-V12.IDL");//Ok
      //f = SD.open("VAPEUR.IDL");//Ok
    auto p = f.position();
    p = p & ~511; // Ensure on a sector boundary, MP3 will resync appropriately
    f.seek(p);
  }

  // // Stuff the buffer with as much as it will take, only doing full sector reads for performance
  while (f && BMP.availableForWrite() > 512) {
    int len = f.read(filebuff, 512);
    BMP.write(filebuff, len);
    if (len != 512) {
      f.close(); // Short reads == EOF
    }
  }

  while(Serial.available()>0)
  {
    char x = Serial.read();
    switch(x)
    {
      case '1': BMP.setGain(0.1);break;
      case '2': BMP.setGain(0.2);break;
      case '3': BMP.setGain(0.4);break;
      case '4': BMP.setGain(0.8);break;
      case '5': BMP.setGain(1);break;
      case '6': BMP.setGain(1.5);break;
      case '7': BMP.setGain(2);break;
    }
  }
```

My file files are just renamed as .IDL but are well wav files.

Is my setup correct for use 16k files , if not what I need to change or add ?
Do i modify my wav files for maximum size, change the buffer ?

Thanks for your help,

I send you my wav file.
[Wav_16K.zip](https://github.com/user-attachments/files/18291285/Wav_16K.zip)

earlephilhower, 2025-01-05:
Please provide a small complete MCVE and single example WAV that's failing.  Your code above has lots of things which don't make sense for WAVs like seeking immediately after opening a file.

pierrotm777, 2025-01-05:
What is complete MCVE.
Sorry my english is bad

earlephilhower, 2025-01-06:
Minimum Complete Verified Example.  Minimize the app and don't give me a random list of files.  If playing A then B then C works, but A then B then D fails, make an app that does that and give just those files.  The idea is to make it easier and foolproof to show the failure outside your own environment so others can look at it without wasting effort figuring out what to do to make it fail.

pierrotm777, 2025-01-06:
The sketch you have have is my code test for this function.
I send you my bad way files this morning.
Can you give me please a better code that mine?
My need is to play a first wav when the motor start and a second when the motor is started.
Thanks 😊

pierrotm777, 2025-01-08:
Sorry for this delay.
I send you my wav files who don't run.
Files are just renamed but are well wav files
[WAV_16K.zip](https://github.com/user-attachments/files/18346136/WAV_16K.zip)

pierrotm777, 2025-01-10:
Sorry again for this delay, I was sick.
This is my Minimum Complete Verified Example .
I send you 2 couple of sounds.
VAPEUR work for .STA and .IDL.
But DSL-LTL have some issue and I don't find why.
[SimpleWAVPlayer.txt](https://github.com/user-attachments/files/18372935/SimpleWAVPlayer.txt)
[WAV_16K.zip](https://github.com/user-attachments/files/18372954/WAV_16K.zip)

Thanks for your help,

Pierre

earlephilhower, 2025-01-21:
Your DSL-LTL files are invalid, that's the problem.  They are 1 byte too long each corrupting the files afterwards.  Fix them in an audio editor (probably just re-save) or force `BMP.flush()` between files to have it throw out any bad data before starting a new WAV.

closed 2025-01-21 by earlephilhower

pierrotm777, 2025-01-22:
Thanks, I see that 👍