earlephilhower/BackgroundAudio #22: MP3 Playback Glitch When Switching to Files with ID3 Tags

Stass48, 2025-01-13:
First, thank you for developing and maintaining the Background Audio library! Its flexibility, including support for various SD libraries like SdFat instead of being tied to SD.h, is a fantastic feature. I hope this independence is preserved in the future.

Now to the issue: When using the library on an RP2040 microcontroller (Raspberry Pi Pico) with Earl Philhower’s core, and outputting audio via I2S to a PCM5102, a problem arises during MP3 file switching. Specifically:
	•	If the library switches to an MP3 file containing ID3 (or other metadata) tags, especially if the tags include embedded cover art (e.g., album artwork), audio artifacts occur.
	•	These artifacts seem to result from the library misinterpreting the ID3 tags, particularly the cover art data, as MP3 audio data, causing incorrect synchronization.
	•	If ID3 tags, including cover art, are removed, switching between files works without artifacts.

Steps to Reproduce:
	1.	Use the Background Audio library with an RP2040 microcontroller and an I2S output to a PCM5102.
	2.	Play any MP3 file.
	3.	Switch to another MP3 file that contains ID3 tags with embedded cover art.
	4.	Observe audio artifacts during switching.

Expected Behavior:
The library should correctly handle MP3 files with ID3 tags, including those with embedded cover art, and switch between files without introducing audio artifacts.

Actual Behavior:
Audio artifacts occur during file switching when MP3 files contain ID3 tags, especially if they include embedded cover art, likely due to the library misinterpreting the tag data as MP3 audio data.

Workaround:
Manually remove ID3 tags, especially embedded cover art, from MP3 files to ensure smooth switching.

Environment:
	•	Microcontroller: Raspberry Pi Pico (RP2040)
	•	Core: Earl Philhower’s RP2040 Arduino Core
	•	Audio Output: I2S via PCM5102
	•	SD Library: SdFat

Again, thank you for this excellent library. Please consider ensuring the library remains flexible and independent of other libraries like SD.h in future updates!

earlephilhower, 2025-01-13:
What is the exact app you're using, and can you post it and a couple of MP3s that cause the issue?

There is no ID3 processing in the library proper, however due to the MP3 framing signals the library will skip anything that's no an MP3 frame.   Throwing ID3s in will just waste its' time a bit.

If you're using the MP3 shuffle example here, then he ID3 parsing and reading can take quite some time due to the special byte encoding (used to avoid including an MP3 frame marker in the ID3 tag).  For text based tags, it's not an issue but for images you might have a 30KB PNG or JPG that needs to both be read out from SD and parsed byte by byte.  Believe it or not, reading that amount off of SD can be pretty painful (until we get that 4-bit interface!) and your artifacts are probably injected silence (clicks).

Increase the background raw data buffer 2x what it is now, and also don't do a `flush` (which isn't necessary...MP3 and AAC both have frame markers) and give it a try.

Stass48, 2025-01-18:
I would like to apologize for my delayed response. I have been trying to understand the issue and how the glitch occurs. Unfortunately, I wasn’t able to reproduce the glitch with other MP3 files. The problem only occurred with these two specific file combinations. It’s possible that the issue lies in the incorrect structure of one of the files or something else. Other files, even those with tags and cover art, played correctly without issues. Of course, when I needed to demonstrate the problem, it worked fine, which was rather ironic :D

However, I did manage to locate the files that caused the issues. These files play fine on a computer player, but I encountered problems with the library and the microcontroller. As requested, I’m attaching the files—both the project file and the MP3 files. Please take a look.

[mp3_demo.zip](https://github.com/user-attachments/files/18464376/mp3_demo.zip)

earlephilhower, 2025-01-18:
Thanks, that's pretty much settled it.

The `2.mp3` has a *153KByte* (decoded) JPEG embedded.  That's just so large that reading it off SD alone (no parsing or de-desynching) just can't be done without underflow.  Your PC can read the whole 8MB into L1 cache at 3GB/s and vector process chunks 16 wide using 8 cores.  The poor little Pico has a 1-bit SPI interface (10MHZ probably!) to the SD card and can't buffer anything tangible.  And it's not just the average time/speed to worry about, because this is real-time decoding the end of the last song in the background while this is going on you have an underflow even if on average all is still well since it can read the actual MP3 data (~16KB/second is all it needs....your image is effectively 10 secs worth of audio!) way faster than needed.

How you got a 300x300 JPG to be 300KB I don't quite know.  😆   The 1.mp3 file's JPG is also 300x300 and only 28KB.  So basically 1/10th the size and time to read/process as the 2nd one.  You can fix the tag in the 2nd file with a shrunk JPG and all should be well...

closed 2025-01-18 by earlephilhower