earlephilhower/BackgroundAudio #27: Examples request, buffer and other questions

Stass48, 2025-01-22:
Hello!
I understand that I may be annoying you with my messages, but I am curious. What if I want to be able to play all available file types that the library can play (mp3, wav, aac) in ONE project, how can I implement this correctly? I did not find such an implementation in the examples. I only found an example with mixing, but I do not need to mix files, I plan to play them one by one. It would be possible to do without a mixer, but I suspect that this will lead to conflicts, since the i2s output will be one for all library subclasses. How can I implement this task correctly? I would like to have more examples of working with the library in future releases.

There are also questions about buffer sizes. In one of the examples I saw a non-standard class declaration:

```
// We will make a larger buffer because SD cards can sometime take a long time to read
BackgroundAudioMP3Class<RawDataBuffer<16 * 1024>> BMP(audio);
```

How do you understand 16*1024? Is it 16 kilobytes? So you could just enter the number 16384 there? Is this the size of the buffer for the audio output? What does it affect?
What values ​​would you recommend for wav and aac?
Maybe you can read about all this somewhere, but I couldn't find it.

Also, I don't really like this section of code:

```
// Stuff the buffer with as much as it will take, only doing full sector reads for performance
while (f && BMP.availableForWrite() > 512) {
int len ​​= f.read(filebuff, 512);
BMP.write(filebuff, len);
if (len != 512) {
f.close(); // Short reads == EOF
}
}
```

Here we first wait to see if the data is needed and only then start preparing it. But we could have prepared the data in advance (read a block from the memory card) and only then wait until this data is needed. In this way, it seems to me, it would have been a little faster. Also, there is a question, why _BMP.availableForWrite() > 512_, why not _BMP.availableForWrite() >= 512_ (in While), the same for _if (len != 512)_, why not _if (len < 512)_. I tried, but I can't write it properly. Any ideas?

Sorry if I'm doing something wrong. Once again, I will say that I am very grateful to you for your work and this library.

Stass48, 2025-01-22:
I also forgot, sorry. How will the code behave if the file size is a multiple of 512?
Maybe we need to add a condition:
if (len > 0) BMP.write(filebuff, len);
What do you think?

earlephilhower, 2025-01-22:
Inline...

> What if I want to be able to play all available file types that the library can play (mp3, wav, aac) in ONE project, how can I implement this correctly? I did not find such an implementation in the [...]

The mixer would be the only thing that talks to the I2S object, so that's definitely one way.  Alternatively just begin and end different decoders on the fly.  Probably would cause some delay and might hit memory fragmentation issues.

> I would like to have more examples of working with the library in future releases.

Magic of open source, feel free to write some and submit pull requests! 😆 

> There are also questions about buffer sizes. In one of the examples I saw a non-standard class declaration:
>
> ```
> // We will make a larger buffer because SD cards can sometime take a long time to [...]

Yes, 16 * 1024 = 16384 so you could put the plain number there.  But if you wanted to try 19KB instead, splitting it up makes it simple to read  `19 * 1024` .  These are [template instantiations](https://en.cppreference.com/w/cpp/language/templates), part of C++.

That's the RawDataBuffer size in bytes.  Pick a number that works for your uses.  No magic formula.  Too large and maybe you'll run out of memory in your app or have very long latencies when changing files mid-stream.  Too small and you'll underflow and get hiccups on the output.

> Also, I don't really like this section of code:
> ```
> // Stuff the buffer with as much as it will take, only doing full sector reads for performance
> while (f && BMP.availableForWrite() > 512) {
> int [...]

I suppose you could read a block ahead, but you've already got 16KB of buffering done in that RDB above.  1/2K isn't going to change things.  The comparison probably could be swapped to `>= 512`, this was an example and test case not fully optimized.  For the exit comparison, `len !=512` will catch any weird error conditions.  Again, I suppose you could look for `<512` but if I asked to read exactly 512b and the file didn't give me exactly 512b, then `read != requested` is what I'd think of first.

> I also forgot, sorry. How will the code behave if the file size is a multiple of 512?
Maybe we need to add a condition:
if (len > 0) BMP.write(filebuff, len);

`BMP.write` should be able to handle 0-byte writes as a no-op so it's not really necessary to check here.  If it can't, then the write method needs to be fixed.

earlephilhower, 2025-01-23:
#32 has real docs and even a GH Doxygen site, so I think we're good here.

closed 2025-01-23 by earlephilhower