earlephilhower/BackgroundAudio #31: It does not even compile

Cyl0nius, 2025-01-23:
C:\Users\tom\Desktop\WebradioMP3PlusWebUI\WebradioMP3PlusWebUI.ino: In function 'void loop()':
C:\Users\tom\Desktop\WebradioMP3PlusWebUI\WebradioMP3PlusWebUI.ino:229:11: error: 'class BackgroundAudioMP3Class<RawDataBuffer<32768> >' has no member named 'pause'
  229 |       mp3.pause();
      |           ^~~~~
C:\Users\tom\Desktop\WebradioMP3PlusWebUI\WebradioMP3PlusWebUI.ino:230:20: error: 'class BackgroundAudioMP3Class<RawDataBuffer<32768> >' has no member named 'paused'
  230 |     } else if (mp3.paused() && mp3.available() > (STREAMBUFF / 2)) { // When paused wait until kind-of full before restarting
      |                    ^~~~~~
C:\Users\tom\Desktop\WebradioMP3PlusWebUI\WebradioMP3PlusWebUI.ino:231:11: error: 'class BackgroundAudioMP3Class<RawDataBuffer<32768> >' has no member named 'unpause'
  231 |       mp3.unpause();
      |           ^~~~~~~

exit status 1

Compilation error: 'class BackgroundAudioMP3Class<RawDataBuffer<32768> >' has no member named 'pause'

earlephilhower, 2025-01-23:
Au contraire, it's in there and CI is actually building everything:

https://github.com/earlephilhower/BackgroundAudio/blob/a7e364ea0680b262521d353898a6bb9319581278/src/BackgroundAudioMP3.h#L135-L145

https://github.com/earlephilhower/BackgroundAudio/actions/runs/12921051880/job/36034387287

Possibly you have a partial copy, with some old and some new files?  A reinstall would be a good first thing to try.

Cyl0nius, 2025-01-23:
Your provided version is 1.1.0 and the arduino-ide is offering a version 1.2.0 ???
I'm confused ...
In version 1.2.0 I can set the i2s parameter, in version 1.1.0 not.

[image]

[image]

[image]

earlephilhower, 2025-01-23:
Ha!  No idea how Arduino library manager picked up 1.2.0 when the release wasn't actually published yet.  Oh well, it's there now but I'd just pull `git master` to avoid the issue in case you've ended up with a weird cached state.

closed 2025-01-23 by earlephilhower