earlephilhower/BackgroundAudio #50: i had run a Ex but play not out after play down

Cans518, 2025-02-26:
this is my code
```cpp
#include <Arduino.h>
#include <BackgroundAudio.h>
#include <LittleFS.h>
#include <PWMAudio.h>
PWMAudio pwm(28);
BackgroundAudioMP3 mp3(pwm);
File f;
uint8_t filebuff[512];
void setup() {
    LittleFS.begin();
    mp3.begin();
    f = LittleFS.open("1.mp3", "r");
    pinMode(25,OUTPUT);
    digitalWrite(25,LOW);
}

void loop() {
    while (f && mp3.availableForWrite() > 512) {
        int len = f.read(filebuff, 512);
        mp3.write(filebuff, len);
        if (len != 512) {
            f.close(); // Short reads == EOF
        }
    }
    mp3.flush();
    digitalWrite(25,HIGH);
}
```
this is my platformio.ini
```txt

Cans518, 2025-02-26:
this is my `platformio.ini`
```txt
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipico2
framework = arduino
board_build.core = earlephilhower
lib_deps = 
	https://github.com/earlephilhower/BackgroundAudio
board_build.filesystem_size = 2m
upload_speed = 921600
build_flags = -std=c++23
```
i run it on pico2
i try to use led to show play is done

Cans518, 2025-02-26:
I have a code that works on ESP8266Audio, but after I updated ardunio-pico my playback became noisy, so I'm trying to use the new library, attached is my previous code:
```cpp
#include <Arduino.h>
#include <AudioFileSourceLittleFS.h>
#include <AudioGeneratorMP3.h>
#include <AudioOutputI2SNoDAC.h>
#include <AudioOutputPWM.h>
#include <map>
#define map std::map // F**K ARDUINO
#include <string>
using std::string;
#include <vector>
using std::vector;

AudioGeneratorMP3 *mp3;
AudioOutputPWM *out;
AudioFileSourceLittleFS *file;

constexpr int internal_led = 25;
constexpr int external_led = 21;
constexpr int button_pin = 18;
constexpr int music_pin = 28;
constexpr int music_en = 26;
constexpr int adc_pin = 27;

const int motor_a = 7, motor_b = 8;
;

# define V_MAX 6

String _d;
string d;

bool debugging = false;

const vector<string> ids_source = {
    {0xc8, 0xf0, 0xbd, 0xf0}, 
    {0xcd, 0xbb, 0xc6, 0xc6, 0xb5, 0xda, 0xc8, 0xfd, 0xb5, 0xc0, 0xb7, 0xe2, 0xcb, 0xf8, 0xcf, 0xdf}, 
    {0xd1, 0xaa, 0xd5, 0xbd, 0xcf, 0xe6, 0xbd, 0xad}, 
    {0xc7, 0xbf, 0xb6, 0xc9, 0xce, 0xda, 0xbd, 0xad}, 
    {0xd5, 0xbc, 0xc1, 0xec, 0xd7, 0xf1, 0xd2, 0xe5}, 
    {0xcb, 0xc4, 0xb6, 0xc9, 0xb3, 0xe0, 0xcb, 0xae}, 
    {0xc7, 0xc9, 0xb6, 0xc9, 0xbd, 0xf0, 0xc9, 0xb3, 0xbd, 0xad}, 
    {0xc7, 0xbf, 0xb6, 0xc9, 0xb4, 0xf3, 0xb6, 0xc9, 0xba, 0xd3}, 
    {0xb7, 0xc9, 0xb6, 0xe1, 0xe3, 0xf2, 0xb6, 0xa8, 0xc7, 0xc5}, 
    {0xc5, 0xc0, 0xd1, 0xa9, 0xc9, 0xbd}, 
    {0xed, 0xae, 0xb9, 0xa6, 0xbb, 0xe1, 0xca, 0xa6}, 
    {0xb9, 0xfd, 0xb2, 0xdd, 0xb5, 0xd8}, 
    {0xbc, 0xa4, 0xd5, 0xbd, 0xc0, 0xb0, 0xd7, 0xd3, 0xbf, 0xda}, 
    {0xbb, 0xe1, 0xc4, 0xfe, 0xb4, 0xf3, 0xbb, 0xe1, 0xca, 0xa6}, 
    {0xce, 0xe2, 0xc6, 0xf0, 0xd5, 0xf2, 0xbb, 0xe1, 0xd2, 0xe9}, 
    {0xd1, 0xd3, 0xb0, 0xb2}, 
};
vector<string> ids;
const vector<string> names_source = {
    "/1.mp3",
    "/2.mp3",
    "/3.mp3",
    "/4.mp3",
    "/5.mp3",
    "/6.mp3",
    "/7.mp3",
    "/8.mp3",
    "/9.mp3",
    "/10.mp3",
    "/11.mp3",
    "/12.mp3",
    "/13.mp3",
    "/14.mp3",
    "/15.mp3",
[... 205 lines cut ...]
    }
  }
  while (mp3->isRunning()) {
    if (!mp3->loop()) {
      mp3->stop(); 
      delete file;
      file = nullptr;
    }
    if (digitalRead(23) == LOW) {
      button_count++;
      if (button_count == 10) {
        mp3->stop();
        button_count = 0;
        delete file;
        curfile++;
        if (curfile >= 2) curfile = 0;
        file = new AudioFileSourceLittleFS(filename[curfile].c_str());
        mp3->begin(file, out);
      }
    }
  }
  delay(1);
}
````
Perhaps this should be reported in the ESP8266Audio library, but I would still like to be able to provide a complete problem description for this issue

earlephilhower, 2025-02-26:
The code in the initial post will not actually do anything because you `mp3.flush()` which throws away any data you ever wrote.  Don't call that unless you want to immediately stop the sound.

And if the MP3 is larger than the small compressed buffer, the code in the 1st post will set the LED high even before the MP3 file has completely been written (i.e. mp3.available will be 0 but f will not be fully read).  Please review the examples like PlayAACSD for more info.

closed 2025-02-26 by earlephilhower