earlephilhower/BackgroundAudio #51: Pico 2 AudioOutputBase.h missing?

relwin, 2025-03-06:
In compiling BeepSpeech/BeepWAV examples I get:
Arduino\libraries\BackgroundAudio\src/WrappedAudioOutputBase.h:49:10: fatal error: AudioOutputBase.h: No such file or directory

Arduino IDE 2.3.4
Pi Pico 2 dev board
BackgroundAudio installed as a ZIP Library.

Am I missing a lib?
Thanks!
-r

earlephilhower, 2025-03-06:
Update your arduino-pico core.  That file was added several releases ago and it looks like you're running a version from before that addition.

closed 2025-03-06 by earlephilhower

relwin, 2025-03-08:
Ok, this works! The IDE's board manager was stuck on version 4.3.1. After a manual Git update it's now listing all versions to 4.4.4.
Thanks for porting this to the Pico.
-r