earlephilhower/BackgroundAudio #55: Please add wiring diagrams

s-c-p, 2025-04-20:
Hi, please add which components should be used, and wiring diagrams for each of the examples

s-c-p, 2025-04-20:
It is okay if you just upload a photo of the real circuit (I guess you must have tested it out)
I will make SVG diagram and send a PR with readme for each example folder

earlephilhower, 2025-04-20:
Thanks for the offer.  I didn't include any wiring diagrams because there's nothing special about this library vs. whatever you'd hook up for audio output using the respective Arduino cores.

For example, the WebRadio-over-PWM example is really just any 2 consecutive pins and GND to a 3.5mm female jack.  And I2S on the RP2040 can pretty much be any I/O (I think it's the same on the later ESP32 models, they have really powerful output muxes).

closed 2025-05-13 by earlephilhower