earlephilhower/BackgroundAudio #57: cannot open source file "driver/i2s_std.h" (dependency of "ESP32I2SAudio.h")  - ESP32 Implementation with VScode/PlatformIO

SondaBerlin, 2025-05-04:
Platform is an ESP32-S3. I am trying to use the Example Mixer.ino: It works when compiling with Arduino IDE. However, when using a more professional IDE like VSCode/PlatformIO I cannon make it run.

Problem: cannot open source file "driver/i2s_std.h" (dependency of "ESP32I2SAudio.h")

Anyone experiencing this? It would be nice to work with this library in VSCode

earlephilhower, 2025-05-04:
That file is part of the IDF newer API that's used in the Arduino core.  Is there a version pin or update you can do in your INI for however you're getting the IDF in your P.IO build?  It's evidently either not using the version shipped with the Arduino core or not including the proper paths for the new API drivers.

earlephilhower, 2025-05-04:
You might also want to post something on the [P.IO community forum](https://community.platformio.org/).  This might be a common issue for other apps as well.

earlephilhower, 2025-05-12:
Looks like the Platform.IO Arduino framework `framework-arduinoespressif32` that's shipped for Platform.IO is not the same as the actual releases of the Arduino core.  It's still on a much older 4.x IDF:

Platform.IO `framework-arduinoespressif32/versions.txt `
````
esp-idf: v4.4.7 38eeba213a
arduino: idf-38eeba213a d75795f5
esp-dl: master 0632d24
esp-rainmaker: master d8e9345
esp32-camera: master f0bb429
esp_littlefs: master 41873c2
espressif__esp-dsp: master 9b4a8b4
tinyusb: master a0e5626bc
````
whereas this library and the later Arduino cores need IDF 5.x. as in esp32-arduino-libs
````
esp-idf: release/v5.4 2f7dcd862a
arduino: idf-release/v5.4 5211d1fa
tinyusb: master 8c1802e41
chmorgan__esp-libhelix-mp3: 1.0.3
espressif__cbor: 0.6.0~1
espressif__esp-dsp: 1.4.12
....
````

Maybe you can ask Espressif or Platform.IO to move to the current release?  I'm sure this isn't the only app that needs a modern IDF.

That said, there's nothing we can do here unless a P.IO expert like @maxgerhardt or someone @ Espressif has any suggestions.

maxgerhardt, 2025-05-13:
>Maybe you can ask Espressif or Platform.IO to move to the current release? 

Sadly theye're PlatformIO and Espressif are in pretty much the same position as PlatformIO + RaspberryPi: Deadlocked, no current updates beyond Arduino-ESP32 2.x. The lore on that can e.g. be found in https://github.com/platformio/platform-espressif32/issues/1225.

However, Jason2866 is doing a fantastic job at supporting Arduino-ESP32 3.x at [pioarduino](https://github.com/pioarduino/platform-espressif32).

All you have to do to use it is to change the `platform` in the `platformio.ini` to one of those releases, for example

```ini
; Has Arduino-ESP32 3.2.0
platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip
```

This should make this library compile as normal.

earlephilhower, 2025-05-13:
Much appreciate the info!  Seems like an "own goal" for both sides again, sadly.  I'll note the community ESP32 P.IO integration in the readme.

earlephilhower, 2025-05-13:
Verified the compile does succeed with `pioardino`.  Thanks again!

PR #60 (2025-05-13): Update readme with community PIO-ESP32 info
Fixes #57
commit 4ca23a7e (2025-05-13): Update readme with community PIO-ESP32 info (#60)

Fixes #57
closed 2025-05-13 by earlephilhower