earlephilhower/BackgroundAudio #63: Feature Request: Add support for PWM and DAC audio output on ESP32 platform

xinaovo, 2025-07-23:
Hi there. Like the title says, ESP32 can generate PWM signal using its LEDC peripheral. Espressif also supports audio output via LEDC peripheral or DAC peripheral in their official SDK. So I'm wondering if these features could be supported in this library. Some useful(maybe) links are shown below:
[https://docs.espressif.com/projects/esp-iot-solution/en/latest/audio/pwm_audio.html](https://docs.espressif.com/projects/esp-iot-solution/en/latest/audio/pwm_audio.html)
[https://docs.espressif.com/projects/esp-iot-solution/en/latest/audio/dac_audio.html](https://docs.espressif.com/projects/esp-iot-solution/en/latest/audio/dac_audio.html)

earlephilhower, 2025-07-24:
For the DAC, I think we looked into it on the ESP8266Audio repo recently.  Isn't it supported only the original, very long in the tooth, ESP32?  And I think it only was ~8bit effective resolution, (even though it takes 16-bit I2S, the DAC portion isn't accurate to 1/65536, only 1/256)?

For PWM, do you know what devices support the LEDC?  It's hard to keep track of what is supported where with so many different SKUs in active production...

xinaovo, 2025-07-30:
By the way, someone on this Chinese video site tried to play music with the built-in DAC, the link is shown below. To be honest, the audio quality was not bad. It should be decent for playing voice.
[https://www.bilibili.com/video/BV18P4y157WV](https://www.bilibili.com/video/BV18P4y157WV)

earlephilhower, 2025-08-12:
Closing for now because I don't see a way forward for the PWM (IRQ, non-DMA) based and because the DAC is a very rare beast.  We have PDM support now (1-bit digital for use with an LPF).  I'd be fine looking at a PR implementing them if you're up to it, but it's not something I think I'll be actively working on.  Thanks, though, for the information and idea!