earlephilhower/BackgroundAudio #64: I2S can be used to generate PDM signals on single DOUT pin without needing CLK pin!

Meshwa428, 2025-08-10:
I have an esp32 S3 and i was experimenting with it to get audio out from single gpio pin because all of my gpios are hogged up by other modules.
And thats when i stumbled across a YT vid, and there was a blog pose on atomic14 also.
and they said we can use PDM signals to output audio and they didnt even use the CLK pin.

can you add it to this project ?

here is the github which has the source [code](https://github.com/atomic14/esp32-pdm-audio)

earlephilhower, 2025-08-10:
I know that code says it's PDM, but I don't think it actually implements it.  All the code there does, that I can see, is just amplify the PCM and clip.  For PDM, AIUI, you'd need to have a bit by bit with accumulate process and the output values would be almost random when looked at like integers, not a scaled representation.

That is a nice trick, though, to assign the unnecessary I2S pins to unused pins!

I did implement a Sigma Delta 1-bit (PDM with configurable oversampling) output in [ESP8266Audio](https://github.com/earlephilhower/ESP8266Audio/blob/master/src/AudioOutputI2SNoDAC.cpp) which could be added here.  Since I moved to the Raspberry Pi Pico cores, and the have nice DMA based PWM, I just never saw the need to pull it in.

Meshwa428, 2025-08-11:
Hey, thank you for the response. I figured it out how to abuse the I2S to provide audio output on a single gpio pin.
you can check this issue: https://github.com/earlephilhower/ESP8266Audio/issues/757

earlephilhower, 2025-08-12:
I tried throwing something together in #68 for this and am running the WebRadio example over PDM.  But, like you, I'm getting a very strange audio output.  It's not something simple like sending stereo 16:16b where it's expecting mono 16b AFAICT.  Might be related to the upscaling the PDM unit uses bu default...

PR #68 (2025-08-12): Add 1-bit PDM output for ESP32
Strange frequency artifacts still seem to appear.  Maybe a more experienced ESP32 PDM developer can take a look...

Fixes #64
closed 2025-08-12 by earlephilhower